ZerBea/hcxdumptool #491: Only one client appear

silan10, 2025-01-10:
I have a AP above i knew sure there is more than 3 devices active but when i filter bpf and start tool only one show

That is command i used

`sudo systemctl stop Networkmanager services
sudo hcxdumptool -m wlan1
sudo hcxdumptool -i wlan1 -F --rcascan=active
sudo hcxdumptool --bpfc="wlan addr3 a000000000000" > filter.bpf
sudo hcxdumptool -i wlan1 -w filter.pcapng --bpf=filter.bpf -F --rds=1 `

silan10, 2025-01-10:
I used aircrack and got handshake quick but in hcxdumptool
I get + only in 1 3 S  
R P  nothing

ZerBea, 2025-01-10:
First of all, this is not an issue, because hcxdumptool is working as expected.
I move this report to discussions, where it belongs