ZerBea/hcxdumptool #517: Why --help doesn't show usage?

fa1rid, 2025-07-12:
hcxdumptool --help
hcxdumptool 6.3.5  (C) 2025 ZeroBeat
Additional information:
-----------------------
press GPIO button to terminate
 hardware modification is necessary, read more:
 https://github.com/ZerBea/hcxdumptool/tree/master/docs
to store entire traffic, run tshark in parallel on the same interface:
 $ tshark -i <interface> -w allframes.pcapng

Berkeley Packet Filter:
-----------------------
  see man bpfc

Important recommendation:
-------------------------
Do not set monitor mode by third party tools or third party scripts!
Do not use virtual interfaces (monx, wlanxmon, prismx, ...)!
Do not use virtual machines or emulators!
Do not run other tools that take access to the interface in parallel (except: tshark, wireshark, tcpdump)!
Do not use tools to change the virtual MAC (like macchanger)!
Do not merge (pcapng) dump files, because this destroys assigned hash values!

fa1rid, 2025-07-12:
Also, I was trying to run it on RPI 3B+, the latest git version seems broken. It will get stuck at `starting...`
I tried the last released version 6.3.5 instead and it worked. What's going on with the master git?
I find this tool unstable since the major refactor you did. It used to be better before the refactor and able to get more handshakes.

```
hcxdumptool --help
hcxdumptool 6.3.5  (C) 2024 ZeroBeat
Additional information:
-----------------------
get information about running services that have access to the device:
 $ sudo systemctl --type=service --state=running
stop all services that have access to the interface, e.g.:
 $ sudo systemctl stop NetworkManager.service
 $ sudo systemctl stop wpa_supplicant.service
run hcxdumptool - it will set an appropriate monitor mode
 scan for ACCESS POINTS in range (packets are not stored to dump file, not in combination with attack modes)
  $ hcxdumptool -i INTERFACENAME -F --rcascan=active
 attack target(s) (not in combination with rcascan)
  $ hcxdumptool -i INTERFACENAME -w dumpfile.pcapng -F --rds=1
   i     : name of the interface to be used
   w     : name of file to which packets are written
   F     : use all available channels
   rds=1 : sort real time display by status (last PMKID/EAPOL on top)
press ctrl+c to terminate
press GPIO button to terminate
 hardware modification is necessary, read more:
 https://github.com/ZerBea/hcxdumptool/tree/master/docs
to store entire traffic, run tshark in parallel on the same interface:
 $ tshark -i <interface> -w allframes.pcapng

Berkeley Packet Filter:
-----------------------
tcpdump decimal numper format:
 example: tcpdump high level compiler:
  $ tcpdump -s 1024 -y IEEE802_11_RADIO wlan addr3 112233445566 -ddd > filter.bpf
  see man pcap-filter
 example: bpf_asm low level compiler
  $ bpf_asm filter.asm | tr ',' '\n' > filter.bpf
  see https://www.kernel.org/doc/html/latest/networking/filter.html
 example: bpfc low level compiler:
  $ bpfc -f tcpdump -i filter.asm > filter.bpf
  see man bpfc
tcpdump C style format:
 example: tcpdump high level compiler:
  $ tcpdump -s 1024 -y IEEE802_11_RADIO wlan addr3 112233445566 -dd > filter.bpf
  see man pcap-filter
 example: bpfc low level compiler:
  $ bpfc -f C -i filter.asm > filter.bpf
  see man bpfc

Important recommendation:
-------------------------
Do not set monitor mode by third party tools or third party scripts!
Do not use virtual interfaces (monx, wlanxmon, prismx, ...)!
Do not use virtual machines or emulators!
Do not run other tools that take access to the interface in parallel (except: tshark, wireshark, tcpdump)!
Do not use tools to change the virtual MAC (like macchanger)!
Do not merge (pcapng) dump files, because this destroys assigned hash values!
```

But this `--help` doesn't seem to complete either.

ZerBea, 2025-07-12:
First of all, I see several questions, but not an issue report. So I'm going to move this report to discussions.

As you noticed, hcxdumptool has been refactored since version 6.3.0. I did this due to performance reasons.
This reasons are detailed explained in the changelog:
https://github.com/ZerBea/hcxdumptool/blob/master/changelog

and in the git discussion section, e.g.:
https://github.com/ZerBea/hcxdumptool/discussions/348
https://github.com/ZerBea/hcxdumptool/discussions/503
https://github.com/ZerBea/hcxdumptool/discussions/343

_the latest git version seems broken. It will get stuck at starting..._
No, it isn't. Due to performance reasons (e.g. running headless on a [Raspberry Pi Zero)](https://github.com/ZerBea/hcxdumptool/discussions/478#discussioncomment-11567630) the real time display is disabled by default:
```
--rds=<digit>    : enable real time display
                    attack mode:
                     0 = off(default)
                     1 = show APs on current channel, show CLIENTs (M1M2ROGUE)
                     2 = show all APs (M1M2, M1M2M3 or PMKID), show CLIENTs (M1M2ROGUE)
                     3 = show all APs, show CLIENTs (M1M2ROGUE)
                     columns:
                      E = encryption (e)ncrypted / (o)pen
                      A = AKM (p)re-shared key
                      1 = received M1
                      2 = received M1M2
                      3 = received M1M2M3
                      P = received PMKID
                    rcascan mode
                     0 = show APs on current channel sorted by BEACON timestamp
                     1 = show APs on current channel sorted by PROBERESPONSE timestamp
                     2 = show APs on current channel sorted by RSSI
                     3 = show APs of all channels sorted by RSSI
--rdt            : disable TIOCGWINSZ for real time displays
```

BTW:
By latest commit aI added an information to --help that there is another menu (-h):
```
$ hcxdumptool --help
hcxdumptool 6.3.5-149-g935b518  (C) 2025 ZeroBeat
Additional information (for commonly used options use -h):
----------------------------------------------------------
...
```

BTW2:
You mentioned that you're running a Raspberry3+. I suggest to give OpenWRT 24.10 a try