ZerBea/hcxdumptool #525: Always creates AP (proberespons)

LowSkillDeveloper, 2025-09-05:
I compiled the latest version of hcxdumptool (7.0.0-1-g33ee466).
According to the README, there should be an --attemptclientmax option, but it’s not available anymore.
The closest replacement I found is --m2max, but even when I set it to 0, hcxdumptool still creates access points.

My goal is to run purely passive capture, so I tried this command:

sudo hcxdumptool -i wlan1 --bpf=filter.bpf --associationmax=0 --m2max=0 --disable_disassociation --rds=3 -c 11a

But APs are still created when my smartphone is scanning for networks.
How can I completely disable AP creation in the current version?

I also tried to use the --proberesponsetx parameter (to set it to 0), but the tool reports it as an unrecognized option:

sudo hcxdumptool -i wlan1 --bpf=filter.bpf --associationmax=50 --disable_disassociation --rds=3 -c 1a --proberesponsetx=0
hcxdumptool: unrecognized option '--proberesponsetx=0'
hcxdumptool 7.0.0-1-g33ee466

while in -h, it displays information that the command exists.

LowSkillDeveloper, 2025-09-07:
I meant I want to exclude getting AP-less data. But hcxdumptool still does proberespons.

ZerBea, 2025-09-07:
To prevent active AP-LESS attacks please use m2max=0

I've updated README.md a couple of minutes ago.

ZerBea, 2025-09-08:
Please try latest commit:
https://github.com/ZerBea/hcxdumptool/commit/4ea5c02a17fef8848f3395bf9c09ca2c4ee5647c
```
--proberesponsetx=<digit> : transmit n PROBERESPONSEs from the ESSID ring buffer
                             default: 1023
                             to disable transmission of PROBERESPONSE frames set 0
```

It does not disable to respond to ASSOCIATIONREQUESTs of CLIENTs to retrieve its M2.
```
--m2max=<digit>           : set maximum of received M1M2ROGUE
                             default: 4 M1M2ROGUE
                             to reject CLIENTs set 0
```

If you don't want that hcxdumptool respond to a CLIENT and(!) if you don't want that hcxdumptool respond to an ASSOCIATIONREQUEST of a CLIENT it is mandatory to set both options to 0.

LowSkillDeveloper, 2025-09-09:
$ sudo hcxdumptool -i wlan1 --bpf=filter.bpf --rds=3 --m2max=0 --proberesponsetx=0

hcxdumptool: unrecognized option '--proberesponsetx=0'
hcxdumptool 7.0.0-3-g4ea5c02 (C) 2025 by ZeroBeat
This is a penetration testing tool!
It is made to detect vulnerabilities in your NETWORK mercilessly!

ZerBea, 2025-09-09:
Thanks for reporting that issue.

Should be removed by latest release
```
$ hcxdumptool -v
hcxdumptool 7.0.1 (C) 2025 ZeroBeat
running on Linux kernel 6.16.5-arch1-1
running GNU libc version 2.42
compiled by gcc 15.2.1
compiled with Linux API headers 6.16.0
compiled with GNU libc headers 2.42
enabled BPF compiler

$ sudo hcxdumptool -i wlan1 --bpf=filter.bpf --rds=3 --m2max=0 --proberesponsetx=0

This is a highly experimental penetration testing tool!
It is made to detect vulnerabilities in your NETWORK mercilessly!
Misuse within a network, without specific authorization, may cause
irreparable damage and result in significant consequences!
Not understanding what you were doing is not going to work as an excuse!

starting...
^C
73 Packet(s) captured by kernel
0 Packet(s) dropped by kernel
exit on sigterm
```

closed 2025-09-09 by LowSkillDeveloper